emeurrens/parking-availability-system #28: View video stream on Raspberry Pi through hardwired connection

emeurrens, 2024-10-29:


emeurrens, 2024-11-03:
- [x] Got live video feed from TCP, not over hardwired connection, but good enough.  Recording in OneDrive (Assignments > Design Prototype)

- [x] Changed frame rate of camera and confirmed frame rate of camera manually using counters and physically timing with stopwatch

Some **bugs** we encountered:
- For some reason, things are . . . [blue](https://www.youtube.com/watch?v=68ugkg9RePc). Check the OneDrive to actually see what I mean (Assignments > Design Prototype > .mkv file)
- Had to power cycle Pi multiple times (4-5 times within a 2-3 hour period) because it would frequently hang when testing Python code. Signs point to slow I/O; processes waiting on I/O operations. Suspicion is on the SD card, so I ordered a new SD card where we will reinstall and reconfigure the RPi.

closed 2024-11-03 by emeurrens

tobonev, 2024-12-07:
Update:

Fixed blue hue on live video stream. We have switched over to primarily using Jupyter Notebook as it seems that Visual Studio was playing a role in the Pi needing to be power cycled.